bitcraze/crazyflie2-nrf-firmware #5: Link in

adjavaherian, 2015-05-23:
Readme link https://github.com/bitcraze/crazyflie2-nrf-firmware/blob/master/nrf51_sdk/readme

Should be changed to https://developer.nordicsemi.com/nRF51_SDK/

dbrgn, 2015-08-14:
Or this one, actually: https://developer.nordicsemi.com/nRF51_SDK/nRF51_SDK_v6.x.x/nrf51_sdk_v6_1_0_b2ec2e6.zip

dbrgn, 2015-08-14:
Still doesn't work properly though:

```
$ cd nrf51_sdk
$ unzip nrf51_sdk_v6_1_0_b2ec2e6.zip
$ cd ..
$ make
arm-none-eabi-gcc -mcpu=cortex-m0 -mthumb -DNRF51  -I Include -I Include/gcc -Iinterface -g3 -O0 -Wall -fsingle-precision-constant -ffast-math -DS110=1 -DBLE=1 -DBLE_STACK_SUPPORT_REQD -DNRF51  -Inrf51_sdk/nrf51822/Include/gcc  -Inrf51_sdk/nrf51822/Include/  -Inrf51_sdk/nrf51822/Include/ble/  -Inrf51_sdk/nrf51822/Include/ble/ble_services/  -Is110/s110_nrf51822_7.0.0_API/include -Inrf51_sdk/nrf51822/Include/app_common/  -Inrf51_sdk/nrf51822/Include/sd_common/    -c -o src/ble/ble.o src/ble/ble.c
In file included from src/ble/ble.c:21:0:
nrf51_sdk/nrf51822/Include/app_common/app_error.h:27:23: fatal error: nrf_error.h: No such file or directory
 #include "nrf_error.h"
                       ^
compilation terminated.
<builtin>: recipe for target 'src/ble/ble.o' failed
make: *** [src/ble/ble.o] Error 1
```

ataffanel, 2015-09-07:
Links have been fixed and the script ```tools/build/download_deps.sh``` have been added to automatically fetch the required dependencies.

closed 2015-09-07 by ataffanel