bitcraze/crazyflie2-nrf-firmware #9: First byte gets altered when transmitting raw packts over nrf chip with new versions

MaEtUgR, 2015-10-22:
I use a completely different firmware on the STM32F405 of the CF2 but the original nrf firmware for handling power and wireless communication.

HOW IT USED TO BE: From old versions (21.07.2015) I'm used to just send a RADIO_RAW syslink packet with my content to the nrf chip and receive the exact same content on my computer.

PROBLEM: In the newest version (29.09.2015) This is not the case anymore but the first byte of my telemetry wireless transaction gets altered by the nrf chip.

For example if I periodically send 0xAB as the first byte with the new version I get 0xA3, 0xA7, 0xA3, 0xA7, 0xA3, ... on my PC. The other bytes stay as they should.
If I periodically send 0x05, I receive 0x01, 0x05, 0x01, 0x05, ...
If I periodically send 0x01, I receive 0x05, 0x01, 0x05, 0x01 ...
You get the idea the first byte is just not usable anymore.

CAUSE: I'm not entirely sure but I think the problem is the commit from (08.09.2015) which adds some messages when nothing is sent (which I don't mind as a new feature) but also dangerously alters the first byte of normal packets we want to send on line 137 in esb.c:
txPackets[txq_tail].data[0] = (txPackets[txq_tail].data[0]&0xf3) | curr_down<<2;

I can work with an old version for my project but please consider this as a real issue for future work with this code.

Thanks in advance and greetings from switzerland
Matthias

ataffanel, 2015-10-23:
I have broken one of my own original design assumption here, you are right the radio_raw packet was supposed to behave like a pipe.

What is happening is that in the Crazyflie the CRTP protocol has 2 reserved bit for link usage, I am using these bits to add a counter to packets to guarantee that there is no packet lost. (out of the box uplink is reliable but downlink (copter to ground) will loose packets).

The commit is this one: https://github.com/bitcraze/crazyflie2-nrf-firmware/commit/59e7130618e05e3e7247341036987b08567e7e3d

I also added a backward compatibility feature: the reliable link has to be enabled which will set the variable "has_safelink" to true. But even when disabled I alter the first byte. We could add a "if (has_safelink)" around the line 137, then by default the packets would not be altered anymore. Do you think that would solve your problems?

MaEtUgR, 2015-10-23:
thanks for your answer

yes, I guess you nailed it!
"has_safelink" should default to false (maybe do this explicit) and if has_safelink is false the packets don't get altered (raw packets get forwarded like in a pipe).

Then the only minimal special behaviour, that would reside is:
When flashing the CF2s STM32 with the original firmware and afterwards flashing back other firmware without restarting the nrf, the flag "has_safelink" would stay true until the next reboot.

But that would be very minor, I just thought about it because I use your original STM32 firmware to flash the nrf chip OTA so we don't have to solder debug channels to the nrf chips of all our CF2s... :)

ataffanel, 2015-10-23:
Safelink is disabled by default, I pushed the fix. Thanks for the report.

closed 2015-10-23 by ataffanel

MaEtUgR, 2015-10-23:
I fetched your bugfix, everything works now!

Even the minor concern I had doesn't affect my workflow. I can again use the newest version for my project.

**Thanks** again, I appreciate

ataffanel, 2015-10-26:
You are welcome. I was not really developing with your use-case in mind but now I know :)

commit cf0236b0 (2015-10-23): Closes #9: Don't touch tx pkt if safelink disabled