bitcraze/crazyflie2-nrf-firmware #10: BLE downlink not working

ataffanel, 2015-11-17:
The BLE downlink is not working properly. It is only working if a Crazyradio is connected at the same time as BLE. When the Crazyradio disconnects the downlink stops even though the STM32 continues to send log packets.

ataffanel, 2015-11-17:
Related to bitcraze/crazyflie-firmware#81

glennvdw, 2015-11-23:
Any updates on what might be causing the issue?

ataffanel, 2015-11-24:
Sending back BLE packet was in the wrong "if", it was not sendig packet back if the ESB (Crazyradio) queue was full. This is now fixed and I tested it by:
 - Connecting with the iPhone app
 - Connecting with the Client, the iphone app see downlink packets
 - Disconnecting the client, the downlink packets continues to come (it is the log block setup by the client)

closed 2015-11-24 by ataffanel

glennvdw, 2015-11-24:
It works!

Thank you :D

commit a947e838 (2015-11-24): Fix #10: BLE downlink broken if ESB queue full