bitcraze/crazyflie2-nrf-firmware #17: Disable bluetooth advertising at runtime

ataffanel, 2016-11-16:
The Bluetooth stack is taking over the radio at regular intervals in order to send advertising packets.

There is 3 situation where it would be beneficial to be able to disable BLE advertising at runtime:
 - When connected with Crazyradio, the advertising is inducing lost packets.
 - Since the stack is taking over the CPU and the highest level interrupts, we cannot talk to the one-wire memory when advertising is enable. This prevents us from writing one-wire deck memories when the nrf firmware is compiled with bluetooth enabled.
 - It corrupts result of the radio-test script: https://github.com/bitcraze/crazyflie-lib-python/pull/21

Last time I tried to disable advertising (early in the Crazyflie 2 development), it froze the firmware. Maybe the full stack needs to be cleanly stopped.

yl5006, 2018-12-05:
@ataffanel is there any low cost hardware to control crazyfile2？

ataffanel, 2018-12-06:
@yl5006 can you please ask your question on the Bitcraze forum (https://forum.bitcraze.io), this issue is used to track development and is not the right forum for random questions.

ataffanel, 2021-03-04:
Already fixed!

closed 2021-03-04 by ataffanel