bitcraze/crazyflie2-nrf-firmware #28: Battery voltage measurement broken if more than 1 deck is installed

ataffanel, 2019-02-06:
With version 2019.02, the battery voltage measurement is reported at 0V when 2 decks are installed. This causes the Crazyflie to put 100% thrust on the motors because of battery voltage compensation.

ataffanel, 2019-02-06:
The problem came from a buffer overflow in the one wire memory caching.

When BLE is enabled, the one wire memories are read and cached in RAM at startup. The reason for this is that BLE is taking CPU time and the one wire link implementation requires interrupt to be switched off in order to manage timing.

The size of the OW cache was of 1 memory, and the code that was suppose to protect buffer overflow was not working. This has not been causing any problem so far because the OW variables where last in ram. However the new multiplatform power management was just after the array ```owCache```:
[image]

This means that if more than 2 decks are installed on a Crazyflie, ```has_rfx2411n``` and ```pmConfig``` are overwritten. On version 2019.01 is causes the radio to have a very low range since it disables the PA on Crazyflie 2.0 and on version 2019.02 it also causes the battery measurement and charging to not work properly.

closed 2019-02-06 by ataffanel

apgiorgi, 2019-02-06:
Thank you @ataffanel!