bitcraze/crazyflie2-nrf-firmware #35: USB connected w/o battery does not go into bootloader

knmcguire, 2019-07-02:
When the USB is connected and there is no battery connected as well,  a long press of the power button does not result the CF2 to go into bootloader mode. LED M2 does start blinking with the long press but upon release it will just restart the CF2

tobbeanton, 2020-01-27:
After some digging I found the problem with the booltoader mode when running from USB power only. It is a bug in the crazyflie2-nrf-bootloader that does not enable 500mA USB current limit. The crazyflie2-nrf-bootloader can be rebuilt with
`make clean all USES_RFX2411N=1 USES_TI_CHARGER=1 flash`
which will enable the 500mA mode and fix the problem. Next problem is that the nrf-bootloader only can be updated with SWD. I suggest investigate the possibility to update it with the wireless bootloader.

ataffanel, 2020-01-28:
There is a (currently undocumented) functionality in the MBS that allows to safely flash a new bootloader and (possibly) BLE softdevice over the air. It requires to flash the new bootloader anywhere in flash and then to write a [structure](https://github.com/bitcraze/crazyflie2-nrf-mbs/blob/c723182/src/main.c#L257) at a specific place in flash. At next reboot the MBS will then check the bootloader CRC, flash the new bootloader, verify that the flash is done successfully and clear the structure. If anything fails, the structure is still there and the flashing attempt will restart at next reboot.

This procedure is safe **as long** as the new bootloader is working. So this is not to be used for development but should be OK for deployment of a tested and released bootloader.

Using this functionality likely require change in the bootloader client to handle writing images in different places in flash.

jonasdn, 2021-09-08:
What is the status of this issue? @tobbeanton