bitcraze/crazyflie2-nrf-firmware #36: Power button not working anymore when using current master-branch

salocinx, 2019-08-05:
Hi. I cloned and compiled the current master-branch today. Unfortunately, the power button on the Crazyflie does not work anymore. To be more specific, turning off the CF2 works, but when trying to power on again, nothing happens.

Only work-around is to open and close the power connector in order to power on the CF2. I then re-flashed the CF2 with the release version and it's working again.

knmcguire, 2019-08-06:
Hi!
That is indeed a strange error, thanks for reporting it. Are you using the Crazyflie version 2.0 or 2.1 ? And did you do an over-air-flash or through the debugger?

salocinx, 2019-08-06:
Hi. I am using two Crazyflies 2.1 with Lighthouse and LED decks attached. I compiled both, the Crazyflie firmware and the NRF firmware, with 'make'. Then I zipped the two *.bin files including the adjusted manifest.json. Finally I uploaded the firmware zip file by the PC-Client software over-the-air using Bootloader mode.

Let me know if you need more information or if I shall make same further testing for you. Thanks for your attention to this issue. I guess the radio bug fixes you uploaded will depend on the NRF firmware as well?

knmcguire, 2019-08-07:
I investigated the situation, and you are right! The Crazyflie becomes unresponsive with the latest NRF firmware... I flashed the nrf with 'make cload' though from the terminal but that does not seem to make a difference. I also tried the latest release (tag 2019.02.1) and there it is still okay.

I will try to debug the NRF and hopefully find a fix soon. It would be interesting to know which commit broke this, it could be the latest radio fixes since that is done for the NRF chip. Could you revert back to before those commits and check if the problem still exists?

knmcguire, 2019-08-07:
Small update on the matter. I was able to pinpoint the location in the code where the crazyflie becomes unresponsive. The Crazyflie can be started up by connecting and disconnecting the battery. However, once the botton is pressed, the NRF get's stuck in this while loop:

https://github.com/bitcraze/crazyflie2-nrf-firmware/blob/90ac428df1cbb1c3dae26db0a27cc8cdfc9f5706/src/pm.c#L136

I do not know why yet, because that line of code has been around for 5 years. There must be something else that is going on before that. But I'll change the topic of this issue to explain the situation a bit better

Let's continue the investigation!

knmcguire, 2019-08-07:
Another update! I figured out that the latest commit: 90ac428df1cbb1c3dae26db0a27cc8cdfc9f5706, broke it. 

Getting closer:)

knmcguire, 2019-08-07:
I've got to leave this issue today, so hopefully I'll figure it out this week. In the mean time @salocinx, if you make the nrf firmeware with bluetooth disabled (make BLE=0), you shouldn't have this issue. It is just a temporary fix for now.

salocinx, 2019-08-07:
Hey that's amazing, thanks a lot for your great support! I also tried to pinpoint the issue, but my debugging equipment is currently quite bad. I am waiting for a JTAG debugger device which didn't arrive yet. Without, it's quite cumbersome.

I just tried to compile with BLE=0 and it's working fine with the latest master-branch. That's great for now, since I am currently only using the Crazyradio PA. That lets me continue hunting for radio communication bugs ;-) Thanks!

ntamas, 2019-08-10:
I ran into this issue as well today. @salocinx, how did you manage to revert to the latest release version with your CF2 after the initial (broken) attempt? For me, the radio is not working so I cannot upload the latest official release via the radio, and I cannot update the NRF firmware in DFU mode, only the STM32, which is not the problem here.

I tried rebooting the CF2 with a warm boot using `cfloader -w usb://0`, but that did not seem to work either (no response after printing `Reset to bootloader mode ...`).

knmcguire, 2019-08-12:
@ntamas have you tried the following:

1. disconnecting the battery and usb
2. keep the powerbutton pushed in 
3. connect the battery while the powerbutton is still pushed in until LED M2 starts blinking slowely.

This should be able to put the crazyflie in bootloader mode for sure.

knmcguire, 2019-08-12:
I will look at this issue this week again. @ataffanel is back from a holiday so I can also ask about the latest commit while trying to fix this

ntamas, 2019-08-12:
@knmcguire yes, I have, and then I ran into issue #35 :)

knmcguire, 2019-08-12:
Oh yes, that is true, but this only happens if you do step 3 with the USB cable without a battery. Step 3 with a battery or battery + USB should work (I just double checked it with mine)

ntamas, 2019-08-12:
Okay, thanks, will try that! Just to be clear: once I manage to reboot the CF in bootloader mode, will I have to upload the stock firmware with the radio or via USB?

ataffanel, 2019-08-12:
You need to bootload via radio. Crazyflie cannot be bootloaded via USB.

ataffanel, 2019-08-14:
Closing this issue, it was caused by the fix to #3. I reverted the fix in 90ac428df1cbb1c3d and re-opened #3.

closed 2019-08-14 by ataffanel