bitcraze/crazyflie2-nrf-firmware #42: Documentation of the NRF uart flow control seems missing

gautierhattenberger, 2020-02-03:
When developing a custom code for the F4 side of the Crazyflie 2.1 (support for Paparazzi UAV system), I struggled a bit with the communication at 1M baud between the NRF and the F4. I later realized that uart hardware flow control is enabled on the NRF side, but on the F4 side, this flow control should be done by hand (no hardware flow control for uart6).
Also, not doing it properly leads to several transmission errors making the communication with the Crazyradio not reliable.
Unless I missed this part of the documentation, I suggest to add this important information to the developer wiki/doc so that people don't face the sameo) issue in the future.

krichardsson, 2020-02-04:
Hi!

Thanks for the feedback!

> Also, not doing it properly leads to several transmission errors making the communication with the Crazyradio not reliable.

Do you mean that you have seen transmission errors between the STM and NRF in the official Crazyflie firmware?

gautierhattenberger, 2020-02-04:
No, only when using my own firmware on F4 without the (software) flow control.
Funny thing was that it was almost working with debugging options but not with optimized code. It turns out that with debug flags, the code is slower and produce less errors (but only by chance). Using properly the RTS line makes it work without problems anymore.

krichardsson, 2020-02-04:
Great, thanks!