bitcraze/crazyflie2-nrf-firmware #47: nrf - messages timestamps and clock syncing

cld05, 2020-07-22:
Hello,
As I discussed in this forum topic [https://forum.bitcraze.io/viewtopic.php?f=16&t=4350&p=19878#p19878](url) I am interested to use the nrf for relative localization of crazyflies.
The RSSI is not very much relialble so I was thinking to use a time-of-flight approach.
In order to do this, there would be the need to synchronize the nrf's clocks.
I found this topic on the Nordic Semiconductor forum - [https://devzone.nordicsemi.com/f/nordic-q-a/3336/nrf51---time-of-flight-distance?ReplySortBy=CreatedDate&ReplySortOrder=Ascending](url) - which shows that with the nRF52 it should be possible.
I also found some examples for the development of timer synchronization feature here - [https://devzone.nordicsemi.com/nordic/short-range-guides/b/bluetooth-low-energy/posts/wireless-timer-synchronization-among-nrf5-devices](url).

However, I am struggling a bit to understand how and if it is even possible to implement this feature in the nrf firmware for the Crazyflie.
I am having some troubles to understand how actually enter the code in the nrf firmware with the pieces of code developed in the example.
Any hint? Suggestion? Help? 

The example that I posted seems to be relevant for the nrf51 DK but I assume that in principle it should be possible to include it also in the Crazyflie.
Not sure if the requirements are supported though since the time sync example seems to require at least the SDK 14.2.0 while the nrf in the Crazyflie seems to use the SDK 6.10 (if I am correct).

ataffanel, 2020-08-01:
Hi,

As a first though: quickly reading the links I assume you are aware that you will not get nearly the same accuracy between LPS and the NRF radio. At best you can sample the packet arrival at 16MHz while the LPS times packet at 64GHz. Raw measurement using the lps would be in orders of millimeters, here it would be in 10's of meters. I am not sure how useful this will be, but it is for sure an interesting experiment.

I am actually currently looking exactly at that kind of synchronization to implement channel-hopping connections and my suggestion would be for you to look at the nRF51 manual and understand the "peripheral interface", "PPI", "Timer" and "Radio" chapters. The nRF5-series of microcontroller implements a quite peculiar but very powerful inter-peripheral synchronization scheme (called events and tasks) that allows to implement real-time behavior without needing the CPU, this is exactly what you need in your case. Furthermore the radio implements its own state-machine that you need to understand in order to follow the current implementation (it is not very hard but since the radio state-machine runs independently from the CPU, the code ends up being not so explicit).

As for the nRF51DK it should not be a problem: the radio does not require any board-specific setup (other than output power maybe), and everything else is internal setup of the chip. So this kind of code usually end up being mainly board-independent.

I think you will need a timer setup at 16MHz to hold the "timer" in both peer and 2 main functionalities:
1) Either:
   - Schedule the packet transmission at a predefined time. This will likely require a PPI channel to connect one of the TIMER->CC to the RADIO->TXEN or the RADIO->START (depending of your shorts setup). This will allow you to predict the time at which the address is sent and so to transmit it in the same packet.
   - Sample the packet departure time by using a PPI channel from RADIO->ADDRESS to a TIMER->CC capture. This will give you the time at which the address was sent so that you can transmit it to the peer in a later packet.
2) Sample the packet receive timestamp using a PPI channel from RADIO->ADDRESS to a TIMER->CC capture.

This is the same basic functions we have in the LPS radio and should allow you to implement similar ranging algorithm. Not implementing the packet scheduling might allow you to "just" piggy-pack packet timing on top of the existing P2P API and so to be able to more easily implement ranging algorithms in the STM32.

ataffanel, 2020-08-18:
Hi,
Did you make any progress on the matter?

Timing functionality of p2p packets sounds like a nice idea that could be used for many purpose beyond ranging (like synchronizing a swarm clock).

We use Github ticket mainly to track issued that will be integrated into the firmware. Do you plan or wish to have this functionality available into the official firmware? If not, I will consider closing this ticket for now if you do not have any further questions.

cld05, 2020-08-18:
Hello,

Thanks for your message.
So far we did not have much time to work on it.
I came back from vacation one week ago and other matters came on my plate.

I will try to follow up this by this week and come to you about the chances that we have about follwoing this up.
Swarm clock synchronizatio is also another feature which may interest us so I will try to check if we are able to continue working on this issue.

Thanks again.
Claudio

closed 2021-09-08 by jonasdn