bitcraze/crazyflie2-nrf-firmware #88: Communicate battery level over bluetooth

valeriyvan, 2023-02-09:
I'd like to add to iOS app [displaying battery level / warning user about low battery level](https://github.com/bitcraze/crazyflie2-ios-client/issues/39).
What's missing here to make possible sending battery level over bluetooth?

ataffanel, 2023-02-10:
There is currently no good ways to transfer battery level over Bluetooth. The main problem is that this information is available using the Crazyflie's Log subsystem but this subsystem requires to fetch the log's Table of Content (TOC) and this is a pretty heavy operation that would take a lot of time with the current bluetooth speed. The reason the TOC is needed is that the log variables cannot be fetched by name, but by ID, and this ID needs to be discovered.

I see two ways to go forward:
 1. Implement a lighter way to get log variables, either by having an API that returns the ID by name of a variable or by being able to add variable names in a log block. Similar things have already been done in the param subsystem to be able to set/get variable by name to allow the Android client to control decks over bluetooth.
 2. We could implement the official [BLE battery service](https://github.com/bitcraze/crazyflie2-nrf-firmware/blob/91b120d294159704ac8d1cbc35f4eb3556829c45/src/ble/ble.c#L256).

The former would allow to add more functionalities like getting pitch/roll/yaw and other logs. The later might allow to get the battery level to be visible from the OS as well which can be useful, for example, to monitor the charge.

If you are thinking of logging more data in the mobile clients later, we should update the log subsystem. Otherwise both way would look good to me.

valeriyvan, 2023-02-10:
> There is currently no good ways to transfer battery level over Bluetooth. The main problem is that this information is available using the Crazyflie's Log subsystem but this subsystem requires to [...]

I would prefer variant 2 - implementing BLE battery service.

ataffanel, 2023-02-21:
Adding a battery BLE service would work fine for us. If you are interested in looking into it we will gladly accept a pull request, otherwise we will eventually get to it and implement it.

valeriyvan, 2023-02-25:
> Adding a battery BLE service would work fine for us. If you are interested in looking into it we will gladly accept a pull request, otherwise we will eventually get to it and implement it.

I would like to implement this as soon as I have time. If you implement this sooner than me - that's fine.

EMart86, 2025-04-05:
Adding the battery percentage will be an enabler for a start/hover functionality.