bitcraze/crazyflie2-nrf-firmware #89: ds28e05 driver does not write correctly to memory multiplications of 16 bytes

coopper5120, 2023-03-12:
When testing communication with ds28e05, I found following bug:

REPRODUCTION PROCEDURE:
Set addres to write at 0x00. Write 32B of data.

RESULT:
Only 16B is actually written.

EXPECTED RESULT:
Full 32B is written.

PROPOSITION OF SOLUTION:
Function ds28e05WriteMemory():
This:
    if ((length - wBytes) >= (BYTES_PER_PAGE))
instead of this:
    if ((length - wBytes) > (BYTES_PER_PAGE))

This soultion was tested with writing 32B into memory. 
All bytes was actually written.

knmcguire, 2023-03-13:
Thanks for notifying us! We will discuss this at our next triage meeting next week.

knmcguire, 2023-03-20:
@coopper5120 could you perhaps make a PR based on your suggestion? That sounds alright to us.

If you don't have time for that, let us know as well and one of us will do it for you.